robin7331/IKEA-Hackant #8: Can I use esp8266 to directly simulate button?

left4taco, 2019-03-06:
Hi, I'm thinking of using an ESP8266 to control the desk. Its GPIO pin can be set to either 3.3V or 0V. 

Can I connect the pin directly to the button? So that I can set the GPIO to 0 to simulate the button being pushed, and reset it back to 3.3V when I want it to stop. 

Thanks.

tg44, 2019-08-03:
I'm planning to do the same. I'm still searching for a good LIN lib to esp8266, but my idea is to hardwire a wemos D1 mini or a nodeMCU to it. My two concern right now is to find a LIN package to read the data, and if the current electronics can power a nodeMCU or not.

Had you any success?

slaesh, 2020-02-07:
hey there @left4taco and @tg44 .
i already did it, will publish my project soon.
including hardware and so on.. :)

will keep you updated..

[image]

alfonmga, 2020-04-12:
> hey there @left4taco and @tg44 .
> i already did it, will publish my project soon.
> including hardware and so on.. :)
> 
> will keep you updated..
> 
> [image]

Hey @slaesh, did you publish it or not yet? 

I would like to do it soon but  I don't have enough experience with projects like this one so I don't want to fuck it up lol.

BTW did you guys finally tried it? @left4taco @tg44

tg44, 2020-04-12:
Not yet. Still interested in this project, but I had no free-time to possibly brick my working table...

slaesh, 2020-05-10:
@alfonmga @tg44 not yet finished.. covid crashed my plans and results in some urgent projects to be done.. currently waiting for some parts shipped over the sea.. and trying to print a case.. 

hopefully i will be able to tinker in a few weeks! :)

keep you updated!

slaesh, 2020-06-29:
sorry for the long delay..
now created a project on tindie! :) @tg44 @alfonmga 

https://www.tindie.com/products/slaesh/smart-ikea-bekant-controller-replacement-with-oled/

if you are interested, just join the list.. =)

meyerf93, 2020-07-28:
Hi @slaesh what you have done look really good. Could you give more details about the interface of the board ? (restAPI and Mqtt) how do you setup the connection wifi ?

algorni, 2021-05-26:
Does this hack works also for the RODULF standup desk?

CrazyDoge128, 2022-01-28:
> sorry for the long delay.. now created a project on tindie! :) @tg44 @alfonmga
> 
> https://www.tindie.com/products/slaesh/smart-ikea-bekant-controller-replacement-with-oled/
> 
> if you are interested, just join the list.. =)

I'm on the list for so long what's happening? Or could you make one for me? I'm from Czechia.